google-deepmind/mujoco_playground #11: some warnning shappen when i run manipulation.ipynb

kassasin, 2025-01-16:
/home/ubuntu/miniconda3/envs/mujoco_playground/lib/python3.10/site-packages/jax/_src/interpreters/xla.py:132: RuntimeWarning: overflow encountered in cast
  return np.asarray(x, dtypes.canonicalize_dtype(x.dtype))
/home/ubuntu/miniconda3/envs/mujoco_playground/lib/python3.10/site-packages/jax/_src/interpreters/xla.py:132: RuntimeWarning: overflow encountered in cast
  return np.asarray(x, dtypes.canonicalize_dtype(x.dtype))
/home/ubuntu/miniconda3/envs/mujoco_playground/lib/python3.10/site-packages/jax/_src/interpreters/xla.py:132: RuntimeWarning: overflow encountered in cast
  return np.asarray(x, dtypes.canonicalize_dtype(x.dtype))
2025-01-16 18:44:38.791223: E external/xla/xla/service/slow_operation_alarm.cc:73] 
********************************
[Compiling module jit_generate_eval_unroll] Very slow compile? If you want to file a bug, run with envvar XLA_FLAGS=--xla_dump_to=/tmp/foo and attach the results.

kassasin, 2025-01-16:
ps:Locomotion is perfect running on my 4090.

btaba, 2025-01-19:
@kassasin 

I just ran the colab using a colab instance, and can't repro. Does everything train as expected or is JIT hanging for too long?

kassasin, 2025-01-20:
> @kassasin 
> 
> I just ran the colab using a colab instance, and can't repro. Does everything train as expected or is JIT hanging for too long? 

Yes， it has running two days！
I am using 4090 to run this tutorial .

btaba, 2025-01-20:
Ok that's super interesting and also a bummer. I can't repro on 1x RTX 4090. This sounds like it's related to https://github.com/google/brax/issues/569. Will have to open a bug with the JAX team.

Can you try lowering `num_evals_envs` to something like 32 in the brax PPO call for now?

kassasin, 2025-01-20:
> Ok that's super interesting and also a bummer. I can't repro on 1x RTX 4090. This sounds like it's related to https://github.com/google/brax/issues/569. Will have to open a bug with the JAX team.
> 
> Can you try lowering `num_evals_envs` to something like 32 in the brax PPO call for now, to see if that gets rid of the issue?
> 

Ok，I will take a try and report. Thanks!

kassasin, 2025-01-21:
> num_evals_env

Hello, I still meet this problem.
My ppo_params is:
```
action_repeat: 1
batch_size: 512
discounting: 0.97
entropy_cost: 0.02
episode_length: 150
learning_rate: 0.001
network_factory:
  policy_hidden_layer_sizes: !!python/tuple
  - 32
  - 32
  - 32
  - 32
  policy_obs_key: state
  value_hidden_layer_sizes: !!python/tuple
  - 256
  - 256
  - 256
  - 256
  - 256
  value_obs_key: state
normalize_observations: true
num_envs: 32
num_evals: 4
num_minibatches: 32
num_timesteps: 20000000
num_updates_per_batch: 8
reward_scaling: 1.0
unroll_length: 10
```

Elfits, 2025-01-27:
Hi, I have the same problem. I ran it on Colab and my local RTX4090 and had the same problem. On my local RTX4090, there is one more warning about "Very slow compile". 

> ['AlohaSinglePegInsertion', 'PandaPickCube', 'PandaPickCubeOrientation', 'PandaPickCubeCartesian', 'PandaOpenCabinet', 'PandaRobotiqPushCube', 'LeapCubeReorient', 'LeapCubeRotateZAxis']
action_repeat: 1
action_scale: 0.04
ctrl_dt: 0.02
episode_length: 150
reward_config:
  scales:
    box_target: 8.0
    gripper_box: 4.0
    no_floor_collision: 0.25
    robot_target_qpos: 0.3
sim_dt: 0.005

action_repeat: 1
batch_size: 512
discounting: 0.97
entropy_cost: 0.02
episode_length: 150
learning_rate: 0.001
network_factory:
  policy_hidden_layer_sizes: !!python/tuple
  - 32
  - 32
  - 32
  - 32
  policy_obs_key: state
  value_hidden_layer_sizes: !!python/tuple
  - 256
  - 256
  - 256
  - 256
  - 256
  value_obs_key: state
normalize_observations: true
num_envs: 2048
num_evals: 4
num_minibatches: 32
num_timesteps: 20000000
num_updates_per_batch: 8
reward_scaling: 1.0
unroll_length: 10

/home/ /anaconda3/envs/ mjc/lib/python3.13/site-packages/jax/_src/interpreters/xla.py:132: RuntimeWarning: overflow encountered in cast
  return np.asarray(x, dtypes.canonicalize_dtype(x.dtype))
2025-01-27 17:01:38.465476: E external/xla/xla/service/slow_operation_alarm.cc:73] 
********************************
[Compiling module jit_generate_eval_unroll] Very slow compile? If you want to file a bug, run with envvar XLA_FLAGS=--xla_dump_to=/tmp/foo and attach the results.
********************************

btaba, 2025-01-28:
Hi @Elfits and @kassasin , I just tried on my RTX 4090 locally, and I cannot reproduce. I tried multiple environments (PandaRobotiqPushCube, PandaPickCubeOrientation, LeapCubeReorient).

Can y'all give a minimal reproducible example, and ideally open an issue here with the XLA dump: https://github.com/jax-ml/jax/issues using latest JAX version?

btaba, 2025-01-28:
Ok never mind! Was able to repro with `PandaRobotiqPushCube`. Will open a bug and post here.

btaba, 2025-01-28:
Posted here https://github.com/jax-ml/jax/issues/26162

kassasin, 2025-01-29:
> Ok never mind! Was able to repro with `PandaRobotiqPushCube`. Will open a bug and post here. 

Ok, thank you very much .

btaba, 2025-01-29:
Hi @Elfits that warning has been happening for some time and is mostly benign

yun-long, 2025-02-07:
another data point, I am experiencing the same issue when running `PandaRobotiqPushCube ` with RTX 4090.

AvalonGuo, 2025-03-14:
Is this issue fixed? the same warninng when running PandaRobotiqPushCube with RTX 3060
`mujoco==3.2.7
jax==0.5.2
brax==0.12.1`

btaba, 2025-04-08:
@yun-long @AvalonGuo if y'all are experience a slow operation alarm, please open a new bug with some details to reproduce!

https://github.com/jax-ml/jax/issues/26162 fixed the original issue a while ago and the "RuntimeWarning overflow" is benign, so I'm closing the issue for now

closed 2025-04-08 by btaba

kassasin, 2025-04-09:
> @yun-long @AvalonGuo if y'all are experience a slow operation alarm, please open a new bug with some details to reproduce!
> 
> https://github.com/jax-ml/jax/issues/26162 fixed the original issue a while ago and the "RuntimeWarning overflow" is benign, so I'm closing the issue for now

Thanks! It is really helpful. Should I need reinstall some packages?

btaba, 2025-04-09:
@kassasin I believe jax>=0.5.1, OR 0.5.2 to be safe