google-deepmind/mujoco_playground #16: Invalid syntax error

vidurvij, 2025-01-17:
Invalid syntax error on the import line

`Traceback (most recent call last):
  File "/Users/vidurvij/Desktop/MujocoPlayground/try.py", line 3, in <module>
    from mujoco_playground import registry
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco_playground/__init__.py", line 16, in <module>
    from mujoco_playground._src import dm_control_suite
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco_playground/_src/dm_control_suite/__init__.py", line 22, in <module>
    from mujoco_playground._src import mjx_env
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco_playground/_src/mjx_env.py", line 27, in <module>
    from mujoco import mjx
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/mjx/__init__.py", line 18, in <module>
    from mujoco.mjx._src.collision_driver import collision
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/mjx/_src/collision_driver.py", line 47, in <module>
    from mujoco.mjx._src import support
  File "/Users/vidurvij/miniconda3/envs/mujoco/lib/python3.9/site-packages/mujoco/mjx/_src/support.py", line 299
    match spec:`

btaba, 2025-01-17:
Can you please post the entire traceback? It looks like it's pointing to something in MJX rather than Playground. Are you on MuJoCo/MJX versions 3.2.7?

davidmao829, 2025-01-18:
@vidurvij Python 3.9 does not support match statements, you can use a higher version of Python

vidurvij, 2025-01-18:
@btaba @davidmao829 Updating to python 3.10 solved it. You might want to update the documentation.

btaba, 2025-01-19:
Thanks @davidmao829,  we'll update this in mujoco which should also still support 3.9

closed 2025-01-19 by copybara-service[bot]

btaba, 2025-01-19:
For now docs have been updated ^^

davidmao829, 2025-01-20:
> Thanks [@davidmao829](https://github.com/davidmao829), we'll update this in mujoco which should also still support 3.9

Appreciate that

commit 1cd81f3e (2025-01-19): Bump minimum required python version. Fixes #16.

PiperOrigin-RevId: 717086713
Change-Id: I5d11069b34ced8988fdd4897989e1a7650c75350