google-deepmind/mujoco_playground #20: Non-deterministic download of mujoco_menagerie on `mujoco_playground` import

traversaro, 2025-01-17:
Hello, and thanks a lot for the nice work!

I noticed that the library on `import mujoco_playground` downloads the latest commit on the master branch of https://github.com/google-deepmind/mujoco_menagerie . Could it make sense to download a specific commit of `mujoco_menagerie` to ensure that the simulation results do not change over time (without any change in mujoco_playground version)? 

From user experience it would be great if `mujoco_menagerie` was its own PyPI package (to avoid re-downloading it every time a new environment is created), but that is probably much more work.

kevinzakka, 2025-01-17:
That makes a lot of sense @traversaro, I'll push something out soon.

closed 2025-01-17 by copybara-service[bot]

traversaro, 2025-01-17:
Great, thanks!

commit c4f2c13c (2025-01-17): Ensure we check out a specific commit of Menagerie. Fixes #20.

PiperOrigin-RevId: 716748884
Change-Id: I39dbae49c48fe2875b30fc5c0063ca7c5da68701