google-deepmind/mujoco_playground #24: Source code of real-world deployments?

dementrock, 2025-01-19:
Very cool project!

Curious if the source code of the real-world deployments, e.g. the locomotion deployments mentioned in Appendix C.5 based on ros2-control, will also be released?

kevinzakka, 2025-01-25:
Hi @dementrock,

Real world code needs to be cleaned up and properly open sourced. I don’t want to give a concrete date and fall short, so all I can say is yes, we will def open source and I’ll keep this issue open to update it when we do, hopefully soon!

cqu-long, 2025-02-19:
> Hi [@dementrock](https://github.com/dementrock),
> 
> Real world code needs to be cleaned up and properly open sourced. I don’t want to give a concrete date and fall short, so all I can say is yes, we will def open source and I’ll keep this issue open to update it when we do, hopefully soon!

Great job! Thank you for your outstanding contributions to the work.
 I'm very interested in the Leaphand project. The reinforcement learning training in the simulation environment has been completed, and I want to carry out the real-world deployment (sim2real). Could you please give me some good suggestions? When will the code be open-sourced? Thank you for your reply.

erwincoumans, 2025-05-11:
>>Real world code needs to be cleaned up and properly open sourced. 
It has been a while, do you still plan to release the deployment code?

In the meantime, someone else released this:
https://github.com/jloganolson/g1_23dof_basic_locomotion/

btaba, 2026-11-15:
We now ship with deployment code in 1.0  @erwincoumans  (although I didn't do g1, I don't *really* have access to the robot)

closed 2026-11-15 by btaba