google-deepmind/mujoco_playground #30: How does push apply to robot

kassasin, 2025-01-21:
Hello, I am not very familiar with this framework . I want to figure out how push apply to robot in following code:
`    state.info["rng"], push1_rng, push2_rng = jax.random.split(
        state.info["rng"], 3
    )
    push_theta = jax.random.uniform(push1_rng, maxval=2 * jp.pi)
    push_magnitude = jax.random.uniform(
        push2_rng,
        minval=self._config.push_config.magnitude_range[0],
        maxval=self._config.push_config.magnitude_range[1],
    )
    push = jp.array([jp.cos(push_theta), jp.sin(push_theta)])
    push *= (
        jp.mod(state.info["push_step"] + 1, state.info["push_interval_steps"])
        == 0
    )
    push *= self._config.push_config.enable
    qvel = state.data.qvel
    qvel = qvel.at[:2].set(push * push_magnitude + qvel[:2])` 
my question is : 
1. What push_theta means?
2. push_magnitude unit is N?
3. What qvel means? the relationship with push?

kevinzakka, 2025-01-21:
Hi @kassasin, we're sampling a direction in the XY plane, sampling a velocity magnitude in m/s and then applying it to the first two indices of qvel. Those correspond to the first two dimensions of the freejoint, which are the xy coordinates of the body (hence why m/s).

closed 2025-01-21 by kevinzakka

kassasin, 2025-01-22:
> Hi @kassasin, we're sampling a direction in the XY plane, sampling a velocity magnitude in m/s and then applying it to the first two indices of qvel. Those correspond to the first two dimensions of the freejoint, which are the xy coordinates of the root body in this case (hence why m/s). Hope that answers your questions.

Thanks,it is very helpful.How about question 5? Can I apply force to desired link? I think
It is same with velocity?

Rian-Jo, 2025-02-08:
if you want to apply 'force' to the body, you can use [xfrc_applied](https://github.com/google-deepmind/mujoco_playground/blob/0f3adda84f2a2ab55e9d9aaf7311c917518ec25c/mujoco_playground/_src/locomotion/go1/joystick.py#L537C1-L564C19) by changing self._torso_body_id to other body.

kassasin, 2025-02-08:
> if you want to apply 'force' to the body, you can use [xfrc_applied](https://github.com/google-deepmind/mujoco_playground/blob/0f3adda84f2a2ab55e9d9aaf7311c917518ec25c/mujoco_playground/_src/locomotion/go1/joystick.py#L537C1-L564C19) by changing self._torso_body_id to other body.

Thank you, I didn't notice go1's example. Maybe I can applty this to g1's .