google-deepmind/mujoco_playground #34: NaN error

kassasin, 2025-01-24:
hello , I meet the same error with [](https://github.com/google-deepmind/mujoco/issues/1484) . Following is the code that I found related with this error . 
    ``` 
# Floor / foot friction: =U(0.4, 1.0).
    rng, key = jax.random.split(rng)
    friction = jax.random.uniform(key, minval=0.4, maxval=1.0)
    pair_friction = model.pair_friction
    pair_friction = pair_friction.at[0, 0:2].set(friction)
    pair_friction = pair_friction.at[1, 0:2].set(friction)
    # Scale static friction: *U(0.9, 1.1).
    rng, key = jax.random.split(rng)
    frictionloss = model.dof_frictionloss[6:] * jax.random.uniform(
        key, shape=(23,), minval=0.5, maxval=2.0
    )
    dof_frictionloss = model.dof_frictionloss.at[6:].set(frictionloss)

    # Scale armature: *U(1.0, 1.05).
    rng, key = jax.random.split(rng)
    armature = model.dof_armature[6:] * jax.random.uniform(
        key, shape=(23,), minval=0.01, maxval=0.1
    )
    dof_armature = model.dof_armature.at[6:].set(armature)
```

Following code is ok and no error occured.
```
    # Floor / foot friction: =U(0.4, 1.0).
    rng, key = jax.random.split(rng)
    friction = jax.random.uniform(key, minval=0.4, maxval=1.0)
    pair_friction = model.pair_friction
    pair_friction = pair_friction.at[0, 0:2].set(friction)
    pair_friction = pair_friction.at[1, 0:2].set(friction)
    # Scale static friction: *U(0.9, 1.1).
    rng, key = jax.random.split(rng)
    frictionloss = model.dof_frictionloss[6:] * jax.random.uniform(
        key, shape=(23,), minval=0.5, maxval=2.0
    )
    dof_frictionloss = model.dof_frictionloss.at[6:].set(frictionloss)

    # Scale armature: *U(1.0, 1.05).
    rng, key = jax.random.split(rng)
    armature = model.dof_armature[6:] * jax.random.uniform(
        key, shape=(23,), minval=1.0, maxval=1.05
    )
```
The difference is armature range .Can anyone explain for this?

kassasin, 2025-01-24:
Does armature can't be two small or other parts relate with armature ?

btaba, 2025-01-24:
Which environment is this for? You're getting NaNs during training for something at HEAD?

kassasin, 2025-01-24:
I create a new env for my robot.And yes, I met nan when training.

kassasin, 2025-01-24:
> I create a new env for my robot.And yes, I met nan when training.

btaba, 2025-01-24:
So this is a new environment for a different robot, or is a new environment for an existing XML in menagerie or mujoco_playground?

kassasin, 2025-01-24:
> So this is a new environment for a different robot, or is it a new environment for an existing XML in menagerie or mujoco_playground?

A new robot .

btaba, 2025-01-24:
Ok it's fairly common to get NaNs when creating a new XML, futzing with the model params is the right way to. I'll move this to a discussion since it's not really a bug with Playground, but feel free to ask more questions