google-deepmind/mujoco_playground #35: Reset the whole scene include tendons

BaiBaiBaiSTRONG, 2025-01-24:
Hi, I have a need to use tendon in my bot setup. I have learned the reset method for each sub-env in the go1 folder. 

Found that it just resets based on the position and joint information read. But once I need to reset an element containing a tendon, I can't reset the whole flexible rope part because I have no way to get the information about the joints on the tendon from qpos or data. 
```python
def reset(self, rng: jax.Array) -> mjx_env.State:
    # Sample a random initial configuration with some probability.
    rng, key1, key2 = jax.random.split(rng, 3)
    qpos = jp.where(
        jax.random.bernoulli(key1, self._config.drop_from_height_prob),
        self._get_random_qpos(key2),
        self._init_q,
    )

    # Sample a random root velocity.
    rng, key = jax.random.split(rng)
    qvel = jp.zeros(self.mjx_model.nv)
    qvel = qvel.at[0:6].set(
        jax.random.uniform(key, (6,), minval=-0.5, maxval=0.5)
    )

    data = mjx_env.init(self.mjx_model, qpos=qpos, qvel=qvel, ctrl=qpos[7:])

    # Let the robot settle for a few steps.
    data = mjx_env.step(self.mjx_model, data, qpos[7:], self._settle_steps)
    data = data.replace(time=0.0)

```

After checking the documentation and experimenting, one possible way I can successfully reset it is by .
```python
 mj.mj_resetData(self.model, self.data)
```

Is it possible for such an idea can be integrated into a custom training task?

The XML set can be used to test:
```xml

```

btaba, 2025-01-24:
Hi @BaiBaiBaiSTRONG , you want a qpos that doesn't violate the tendon constraints, is that right? FWIU `mj_resetData` resets to qpos0, not to a random qpos.

I think what you might need to do is generate a giant list of qpos that don't violate the tendon constraints, and then pick a random index into that list.

BaiBaiBaiSTRONG, 2025-01-25:
Great Thanks for your quick reply!!

What I want to do is exactly reset the system to the original setup state. Things I am not sure is whether reset qpos0 could also consider the tendon together. I am worried that this will only make the end ball and rod joints back to the original but leave the tendon-like rope at the unreset place.

btaba, 2025-01-28:
You might be looking for [`tendon_length0` ](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html#:~:text=2)%0A%20%20mjtNum*-,tendon_length0,-%3B%20%20%20%20%20%20%20//%20tendon%20length)? Closing the issue for now since this is not really a question about Playground. Feel free to open an in issue in the [MuJoCo](https://github.com/google-deepmind/mujoco) if you're still having trouble.

closed 2025-01-28 by btaba