google-deepmind/mujoco_playground #41: "UnexpectedTracerError" during inference after training, when using learning/train_jax_ppo.py

dementrock, 2025-02-02:
Full command:

```bash
JAX_COMPILATION_CACHE_DIR=/tmp/jax_cache python learning/train_jax_ppo.py   --env_name=G1JoystickFlatTerrain   --use_wandb   --domain_randomization   --seed=1   --num_timesteps=400000000   --num_evals=16   --reward_scaling=0.1   --normalize_observations   --unroll_length=32   --num_minibatches=32   --num_updates_per_batch=5   --discounting=0.98   --learning_rate=1e-4   --entropy_cost=0   --num_envs=16384   --batch_size=1024   --max_grad_norm=1.0   --policy_hidden_layer_sizes=512,256,64   --value_hidden_layer_sizes=256,256,256,256   --policy_obs_key=state   --value_obs_key=privileged_state
```

No modifications have been made to the code.

Running on EC2 g6.16xlarge with an L4 GPU.

Relevant error message:

```
0: reward=-6.888                                                                                                                       
27262976: reward=-2.799                                                                                                                                                                                                                                            
54525952: reward=-2.405                                                                                                                                                   
81788928: reward=-2.163                                                                                                                                                                                                                                            
109051904: reward=-1.882                                                                                                                                                                                       
136314880: reward=-1.799                                                                                                                                                  
163577856: reward=-1.448                                                                                                                                                                                                                                           
190840832: reward=-0.860                                                                                                                                                                                       
218103808: reward=0.583                                                                                                                                                   
245366784: reward=1.336                                                                                                                                                                              
272629760: reward=1.918                                                                           
299892736: reward=2.775                                                                                                                                                                              
327155712: reward=2.694                                                                           
354418688: reward=4.944                                                                                                                                                                              
381681664: reward=5.018                                                                           
408944640: reward=5.689                                                                                                                                                                              
Done training.                                                                                    
Time to JIT compile: 104.33590551605448                                                                                                                                                              
Time to train: 20279.01664167503                                                                                                                                                                     
Starting inference...                                                                                                                                                                                
Traceback (most recent call last):                                                                                                                                                                   
  File "/home/ubuntu/mujoco_playground_workspace/mujoco_playground/learning/train_jax_ppo.py", line 433, in <module>                                                                                                                                                                                                                                                                                      
    app.run(main)                                                                                                                                                                                    
  File "/home/ubuntu/mujoco_playground_workspace/miniconda3/envs/mujoco-playground/lib/python3.11/site-packages/absl/app.py", line 308, in run                                                       
    _run_main(main, args)                                                                                                                                                                            
  File "/home/ubuntu/mujoco_playground_workspace/miniconda3/envs/mujoco-playground/lib/python3.11/site-packages/absl/app.py", line 254, in _run_main                                                 
    sys.exit(main(argv))                                                                                                                                                                             
             ^^^^^^^^^^                                                                                                                                                                                                                                                                                                                                                                                   
  File "/home/ubuntu/mujoco_playground_workspace/mujoco_playground/learning/train_jax_ppo.py", line 393, in main                                                                                                                                                                                                                                                                                          
    state = jit_reset(reset_rng)                                                                                                                                                                                                                                                                                                                                                                          
            ^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                                                                                                                                                                                          
  File "/home/ubuntu/mujoco_playground_workspace/miniconda3/envs/mujoco-playground/lib/python3.11/site-packages/mujoco_playground/_src/locomotion/g1/joystick.py", line 260, in reset                                                                                                                                                                                                                     
    data = mjx_env.init(self.mjx_model, qpos=qpos, qvel=qvel, ctrl=qpos[7:])                                                                                                                         
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                                                                                                                           [... cut ...]           
JAX transformations require that functions explicitly return their outputs, and disallow saving intermediate values to global state.                                                                 
To catch the leak earlier, try setting the environment variable JAX_CHECK_TRACER_LEAKS or using the `jax.checking_leaks` context manager.                                                            
See https://jax.readthedocs.io/en/latest/errors.html#jax.errors.UnexpectedTracerError    
```

willthibault, 2025-02-24:
I have encountered this error as well. Any update on this?

Some additional information I can provide is:
- I am using a RTX 4090 GPU
- MJX = 3.2.7, Brax = 0.12.1
- This error occurs with domain randomization on but not when domain randomization is off
- I am also using the G1 locomotion environment with some custom changes to the environment for my own robot but same training file setup

willthibault, 2025-02-24:
It doesn't exactly explain the error above, but reloading `eval_env` prior to defining the reset and step functions for evaluation helps avoid the error:
```
  # Prepare for evaluation
  eval_env = envs.load(_ENV_NAME.value, config=env_cfg)
  jit_reset = jax.jit(eval_env.reset)
  jit_step = jax.jit(eval_env.step)
```

Is it possible that when domain randomization is added that some unexpected behaviour is occurring to the `eval_env` during training?

btaba, 2025-04-08:
Ah yes @willthibault that is expected! Redefine the env after being used in domain rando

commit bc10b8aa (2025-04-09): Fix #41.
closed 2025-04-09 by copybara-service[bot]