google-deepmind/mujoco_playground #44: Save ppo_params along with env_cfg, and load ppo_params automatically when feeding checkpoint path and setting play_only

dementrock, 2025-02-02:
UX would be improved if:
* ppo_params were automatically saved as json
* load them when feeding checkpoint path and setting play_only
* Also auto set num_timesteps to 0 when play_only

annika-nv, 2025-02-06:
+1 this would be very helpful!

kevinzakka, 2025-02-18:
Thanks @dementrock!

Will address these.

karthyyy, 2025-03-18:
@kevinzakka ,

   Whether it's implemented ? Saw a similar request in

https://github.com/google/brax/issues/438

But the solution is not working for me.

Regards
   Karthik

btaba, 2025-03-31:
Hi @karthyyy did you open a related issue in brax RE https://github.com/google/brax/commit/8526f9a64ee02010615a57a026a7b6aad05cbda0?  We'll eventually bubble up the new model saving/loading API in the training script in playground

btaba, 2025-06-23:
This is all piped through now, sorry for the delay

closed 2025-06-23 by btaba