google-deepmind/mujoco_playground #48: The step counter overflows when using a very large num_timesteps

dementrock, 2025-02-05:
I tried setting num_timesteps to 10,000,000,000 and seems like this broke the checkpointing and W&B logging. The logs look like:

```
0: reward=-9.337
 3 666894336: reward=9.799
 4 1333788672: reward=24.241
 5 2000683008: reward=16.648
 6 -1627389952: reward=33.575
 7 wandb: WARNING Tried to log to step -1627389952 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
 8 -960495616: reward=34.962
 9 wandb: WARNING Tried to log to step -960495616 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
10 -293601280: reward=34.532
11 wandb: WARNING Tried to log to step -293601280 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
12 373293056: reward=32.648
13 wandb: WARNING Tried to log to step 373293056 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
14 1040187392: reward=31.220
15 wandb: WARNING Tried to log to step 1040187392 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
16 1707081728: reward=34.702
17 wandb: WARNING Tried to log to step 1707081728 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
18 -1920991232: reward=37.682
19 wandb: WARNING Tried to log to step -1920991232 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
20 -1254096896: reward=22.399
21 wandb: WARNING Tried to log to step -1254096896 that is less than the current step 2000683008. Steps must be monotonically increasing, so this data will be ignored. See https://wandb.me/define-metric to log data out of order.
```

kassasin, 2025-02-10:
I also meet a similar problem," total steps > num_timesteps" when i set "num_timesteps = 3000000000" .

Hippozhibos, 2025-02-11:
similar problem. I set num_timesteps to much smaller value, like100_000, but the actual training process seems run more than 120000 steps.

kassasin, 2025-02-11:
> similar problem. I set num_timesteps to much smaller value, like100_000, but the actual training process seems run more than 120000 steps.

How about your "unroll_length"?

vincentzhang, 2025-02-12:
It seems to me that the underlying issue is the same as the one described in #52.

In [brax/training/agents/ppo/train.py](https://github.com/google/brax/blob/main/brax/training/agents/ppo/train.py#L698), the total_steps value is derived from TrainingState.env_steps, which is a jax.numpy.ndarray that defaults to int32 on the GPU, hence the overflow. Although it’s possible to convert this to int64, I’m not sure what's the best thing to do here given that this code resides in the brax repository.

Hippozhibos, 2025-02-12:
> > similar problem. I set num_timesteps to much smaller value, like100_000, but the actual training process seems run more than 120000 steps.
> 
> How about your "unroll_length"?

unroll_length=30. I didn't change it. What does it used for by the way?

kassasin, 2025-02-12:
> > > similar problem. I set num_timesteps to much smaller value, like100_000, but the actual training process seems run more than 120000 steps.
> > 
> > 
> > How about your "unroll_length"?
> 
> unroll_length=30. I didn't change it. What does it used for by the way?

Thanks, I just meet a similar problem, in my case, unroll_length=100 .

vincentzhang, 2025-02-13:
Opened an issue in brax: https://github.com/google/brax/issues/578

kevinzakka, 2025-02-14:
This is a known issue and we'd probably have to refactor the logic in brax. We tried switching to int64 a few months ago but jax will ignore it unless 64 bit is enabled globally (which we don't want).

vincentzhang, 2025-02-14:
Thanks for the context @kevinzakka . Yeah, it seems that int64 (still) can only be set globally in jax.

btaba, 2025-04-11:
See https://github.com/google/brax/commit/84b9b4ff3ff3e14e1f42bd3e38f045b12153ca24 for the fix

closed 2025-04-11 by btaba