google-deepmind/mujoco_playground #54: Unitree Go2 Support?

sunsleaf, 2025-02-12:
Hi, I wonder if it is in your plan to support unitree go2 in future? Thanks!

youwyu, 2025-02-12:
I guess this can work, although I did not try.
https://github.com/google-deepmind/mujoco_menagerie/tree/main/unitree_go2

kevinzakka, 2025-02-12:
I made a model and task last week, will try to push it to the repo next week. I unfortunately don't have a go2 so can't verify that it works.

sunsleaf, 2025-02-15:
Hi, I'm aware of the go2 model from mujoco_menagerie. But I was wondering if there is a predefined env for that, including suitable parameters for training, joint angles, etc. I'll try it later after the model is pushed. But thanks for the support!

aatb-ch, 2025-04-10:
hi @kevinzakka, i have a go2 implementation running (tested foot only for now but i can add the full collision), should i make a PR?

kevinzakka, 2025-04-10:
@aatb-ch Go for it!

yepw, 2025-04-13:
Thanks for the PR, @aatb-ch Have you had a chance to test the trained policy on a physical GO2?

aatb-ch, 2025-04-13:
hey @yepw, that's what i m working on atm, will let you know when i get it going.

closed 2026-11-15 by btaba