google-deepmind/mujoco_playground #55: Duplicate noisy_joint_vel in privileged_state of go2 joystick

yepw, 2025-02-13:
Hi, the `privileged_state` of go2 joystick env contains `noisy_joint_vel` ([code](https://github.com/google-deepmind/mujoco_playground/blob/254c38f9c240012518dbe61d1267bc375ff827e4/mujoco_playground/_src/locomotion/go1/joystick.py#L365)). 
However,  `noisy_joint_vel` is already in `state` ([code](https://github.com/google-deepmind/mujoco_playground/blob/254c38f9c240012518dbe61d1267bc375ff827e4/mujoco_playground/_src/locomotion/go1/joystick.py#L348)) and `privileged_state` contains `state`. Is this a typo or on purpose? 

Meanwhile, the  `privileged_state` **go1**'s joystick env has `joint_vel` rather than `noisy_joint_vel` ([code](https://github.com/google-deepmind/mujoco_playground/blob/254c38f9c240012518dbe61d1267bc375ff827e4/mujoco_playground/_src/locomotion/g1/joystick.py#L496)).

kevinzakka, 2025-02-13:
It should be joint_vel, good catch! Will push a fix.

kevinzakka, 2025-02-13:
Note in practice this doesn't really make a difference.

commit 0fb7bbdd (2025-02-13): Small fix to go1 joystick privileged state. Fixes #55.

PiperOrigin-RevId: 726651064
Change-Id: I8742728347c15ad9377cd062646a0e178930d004
closed 2025-02-13 by copybara-service[bot]