google-deepmind/mujoco_playground #56: No splitting of the rng in the reset function of BraxDomainRandomizationVmapWrapper

jonasweihing, 2025-02-14:
Hello,

I am working on domain randomization for my RL Env and stumbled upon the following potential problem in the reset function of the [BraxDomainRandomizationVmapWrapper](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/_src/wrapper.py#L137).

Since the wrapper also vectorizes the environment, my expectation would be that the rng must be split before vmapping the reset of the environment, as is done in the [VmapWrapper](https://github.com/google/brax/blob/main/brax/envs/wrappers/training.py#L58). Without splitting the rng the different domain randomized environments would have the same initial state, which I would say is not intended behavior or is it?

This problem also exists in the [DomainRandomizationVmapWrapper](https://github.com/google/brax/blob/main/brax/envs/wrappers/training.py#L220) in Brax, which I guess the BraxDomainRandomizationVmapWrapper was ported from.

kevinzakka, 2025-02-14:
The RNG is [split](https://github.com/google/brax/blob/main/brax/training/agents/ppo/train.py#L119) in the [`_maybe_wrap_env`](https://github.com/google/brax/blob/main/brax/training/agents/ppo/train.py#L97) function in the PPO train.py script.

kevinzakka, 2025-02-18:
Hi @jonasweihing, I believe I've answered your question so going to close. Let me know if you have followup questions.

closed 2025-02-18 by kevinzakka