google-deepmind/mujoco_playground #60: Unable to run `train_jax_ppo.py` due to missing ALL_ENVS

Nate711, 2025-02-18:
I get the following error when I try using  `train_jax_ppo.py`. 
```
python train_jax_ppo.py --help
Traceback (most recent call last):
  File "/home/nathankau/mujoco_playground/learning/train_jax_ppo.py", line 70, in <module>
    f"Name of the environment. One of {', '.join(registry.ALL_ENVS)}",
                                                 ^^^^^^^^^^^^^^^^^
AttributeError: module 'mujoco_playground._src.registry' has no attribute 'ALL_ENVS'
```

As mentioned in this 3 week old issue https://github.com/google-deepmind/mujoco_playground/issues/39, the commit to blame is https://github.com/google-deepmind/mujoco_playground/commit/ea0754306b389f79c67a52b1ddea988a137ae257

I'd really love if the script was fixed thanks!

kevinzakka, 2025-02-18:
I have a fix for this, will be out tomorrow!

closed 2025-02-18 by kevinzakka

Nate711, 2025-02-27:
FYI there are some colabs like https://colab.research.google.com/github/google-deepmind/mujoco_playground/blob/main/learning/notebooks/manipulation.ipynb#scrollTo=B9T_UVZYLDdM that refer ALL_ENVS variables like registry.manipulation.ALL_ENVS