google-deepmind/mujoco_playground #65: Go1's Getup task is not working; when it turns over, it remains prone

lmtss, 2025-02-22:
Neither using brax nor rsl-rl allows the Go1 to stand up after turning over; it remains prone.

Rian-Jo, 2025-02-26:
for me, i trained several times with checkpoints for Getup task. the robot gets done after few trainings.

lmtss, 2025-02-26:
> for me, i trained several times with checkpoints for Getup task. the robot gets done after few trainings.

Wow, that's awesome! Approximately how many timesteps or iterations in total?

lmtss, 2025-02-27:
> for me, i trained several times with checkpoints for Getup task. the robot gets done after few trainings.

My goodness, I did it! I tried training for 750 million timesteps (the default configuration is 50 million), and it worked. Thank you for sharing; this is very helpful.

kevinzakka, 2025-02-27:
Hi, getup can be trained extremely efficiently. We only do extra passes with a power curriculum to deploy safer policies on hardware. Can you try using [this notebook](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/experimental/learning/getup.ipynb) and tell me if you can reproduce the result?

Rian-Jo, 2025-02-27:
@kevinzakka 

i have tested [fall recovery](https://github.com/Rian-Jo/mujoco_playground/blob/e89e0b12d3e7366dded007308d7e92f7d6a1e98e/mujoco_playground/experimental/learning/getup2.ipynb) task. 

https://github.com/user-attachments/assets/d93e8c29-3533-4988-b4bf-6852b6d3b571

i just added installation parts from the notebook you shared and ran training on colab. is there something wrong?

lmtss, 2025-02-27:
> Hi, getup can be trained extremely efficiently. We only do extra passes with a power curriculum to deploy safer policies on hardware. Can you try using [this [...]

I trained getup directly by running `train_jax_ppo.py` instead of using the notebook. Could this be the source of the problem? I can try running the notebook to see if it reproduces the result.

xander-2077, 2025-05-05:
Hello! Can you show me the reward curve and the video of the training effect? @lmtss  ​​I am not sure about my training results. In addition, the effect of [this video](https://github.com/google-deepmind/mujoco_playground/issues/65#issuecomment-2687153202) looks quite different from that in https://playground.mujoco.org/, which seems unreasonable. Have you tried to transfer it to a real robot?

closed 2026-11-15 by btaba