google-deepmind/mujoco_playground #66: Registering new environments to train with existing scripts

Balint-H, 2025-02-25:
Hello!

I'm working on a new MJX environment, but would like to reuse as much of the mujoco playground implementations and practices as possible. I find it tricky to actually register a new environment and then train with `train_jax_ppo.py`. This is because `registry.get_default_config()` checks only the environments defined originally in mujoco playgrounds (i.e. it checks against the `ALL`  dict in the subregistries, insteaf of `_cfg` which contains the more up to date dict). 

Therefore, even if I register a new env into one of the subregistries (as the main one doesn't expose a register function) then it is ignored by the training scripts.

What would be the intended usage of train scripts with new envs? Of course, via custom versions of the script we can access the target environment, but a general use tool would be nice.

vincentzhang, 2025-03-01:
Thanks for opening the issue. I've just started working on a similar use case (a locomotion env). 

My plan was to add the new env under _src/locomotion directory and update [_envs](https://github.com/google-deepmind/mujoco_playground/blob/609168a02cc068193f0fdb4379c2030b388b073e/mujoco_playground/_src/locomotion/__init__.py#L43), [_cfgs](https://github.com/google-deepmind/mujoco_playground/blob/609168a02cc068193f0fdb4379c2030b388b073e/mujoco_playground/_src/locomotion/__init__.py#L84), and [_randomizer](https://github.com/google-deepmind/mujoco_playground/blob/609168a02cc068193f0fdb4379c2030b388b073e/mujoco_playground/_src/locomotion/__init__.py#L109) in [`_src/locomotion/__init__.py`](https://github.com/google-deepmind/mujoco_playground/blob/609168a02cc068193f0fdb4379c2030b388b073e/mujoco_playground/_src/locomotion/__init__.py).
If the new environment doesn't fit under locomotion or manipulation, I guess we could make a new directory under _src/, e.g. _src/thirdparty, _src/custom etc. and fill in the `_src/custom/__init__.py` and update all methods in [_src/registry.py](https://github.com/google-deepmind/mujoco_playground/tree/609168a02cc068193f0fdb4379c2030b388b073e/mujoco_playground/_src/registry.py)
to  include this new directory.

Would that sound like a reasonable plan to address your use case?

Balint-H, 2025-03-04:
I think a thirdparty/custom/user (I'd go with "user") registry would be best, and to expose functioning registration methods for the existing registries (with an optional argument to supply a randomizer too). It sounds simple to do, and it wouldn't affect any of the existing environments. I'd be happy to have a go at implementing it and to send a PR!

vincentzhang, 2025-03-04:
That sounds great to me. Thanks!

Balint-H, 2025-03-05:
I realised that on HEAD this issue is already mostly resolved (registering envs in existing registries are listed correctly), just not in the pip installed package. I'll still consider if there is a point to do custom subregistries. I think there's a potential way to do it with ConfigDict that would allow the user to define whatever naming convention they want to use, while being fully compatible with current environments.

closed 2025-03-05 by Balint-H

karthyyy, 2025-03-18:
Hi @Balint-H ,

   Were you able to add custom environment to playground ??

Regards
   Karthik

Balint-H, 2025-03-18:
Hello,
Yes! If your pip installed version still has the older version of the code, its just one extra line of code. Also, there isn't a generic or custom environment registry yet (I'm still interested in implementing it, just not got around to it), so you have to pick one of manipulation or locomotion registries.

To register your environment, you just need to call this code:
``` python
registry.locomotion.register_environment("MyEnvironmentName", MyEnvironmentClass, default_config_of_my_environment)
registry.locomotion.ALL.append("MyEnvironmentName")
```

You could put this at the start of the main function of e.g. `train_jax_ppo.py`, or put it in an ``__init__.py`` file and import that module (e.g. custom environments) at the start of your training script.

karthyyy, 2025-03-19:
@Balint-H ,

   Thanks. I will try it and get back.

Regards
    Karthik

karthyyy, 2025-03-20:
Hi @Balint-H ,

I tried the same in a Jupyter Notebook. My 'mujoco' version is 3.3.0 & 'playground' version is 0.0.4 (both installed using pip 2-3 weeks back). When I execute the following code.

```
class MyEnvironmentClass:
    def __init__(self, config):
        self.config = config        

    def reset(self):        
        pass

    def step(self, action):        
        pass

default_config_of_my_environment = {
    'param1': 'value1',
    'param2': 'value2',    
}

# Register the environment
registry.locomotion.register_environment("MyEnvironmentName", MyEnvironmentClass, default_config_of_my_environment)

# Add the environment to the list of all environments
registry.locomotion.ALL.append("MyEnvironmentName")
```

   It returns error

"

```
---> 20 registry.locomotion.ALL.append("MyEnvironmentName")

File [~\env_playground\lib\site-packages\mujoco_playground\_src\locomotion\__init__.py:131](http://localhost:8888/lab/tree/~/env_playground/lib/site-packages/mujoco_playground/_src/locomotion/__init__.py#line=130), in __getattr__(name)
    129 if name == "ALL_ENVS":
    130   return tuple(_envs.keys())
--> 131 raise AttributeError(f"module '{__name__}' has no attribute '{name}'")

AttributeError: module 'mujoco_playground._src.locomotion' has no attribute 'ALL'
```
"
  Also on executing 
 
`env = registry.locomotion.make("MyEnvironmentName")`
 
it returns
 
```
----> 1 env = registry.locomotion.make("MyEnvironmentName")

File [~\env_playground\lib\site-packages\mujoco_playground\_src\locomotion\__init__.py:131](http://localhost:8888/lab/tree/~/env_playground/lib/site-packages/mujoco_playground/_src/locomotion/__init__.py#line=130), in __getattr__(name)
    129 if name == "ALL_ENVS":
    130   return tuple(_envs.keys())
--> 131 raise AttributeError(f"module '{__name__}' has no attribute '{name}'")

AttributeError: module 'mujoco_playground._src.locomotion' has no attribute 'make'
```

Regards
   Karthik