google-deepmind/mujoco_playground #68: training_vision_1 tutorial: AttributeError: 'Balance' object has no attribute 'renderer'

benbenwsh, 2025-02-26:
When this line is run in either of the training_vision_1 tutorials:
`state = jit_reset(jax.random.split(jax.random.PRNGKey(0), num_envs))`

This error is raised:
`---------------------------------------------------------------------------

AttributeError                            Traceback (most recent call last)

[<ipython-input-9-fb57eebb09e8>](https://localhost:8080/#) in <cell line: 0>()
      3 
      4 
----> 5 state = jit_reset(jax.random.split(jax.random.PRNGKey(0), num_envs))
      6 rollout = [unvmap(state)]
      7 

    [... skipping hidden 11 frame]

5 frames

    [... skipping hidden 3 frame]

[/usr/local/lib/python3.11/dist-packages/mujoco_playground/_src/dm_control_suite/cartpole.py](https://localhost:8080/#) in reset(self, rng)
    179     obs = self._get_obs(data, info)
    180     if self._vision:
--> 181       render_token, rgb, _ = self.renderer.init(data, self._mjx_model)
    182       info.update({"render_token": render_token})
    183       obs = _rgba_to_grayscale(rgb[0].astype(jp.float32)) / 255.0

AttributeError: 'Balance' object has no attribute 'renderer'`

Andrew-Luo1, 2025-03-08:
Thanks for your interest in the tutorial! In my experience, this appears when the Madrona MJX installation isn't correct. Have you tried
```bash
$ python
Python 3.10.16 (main, Dec 11 2024, 16:24:50) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import madrona_mjx
```
?

For an installation reference, it may help to look at the first cell of the [colab version](https://github.com/google-deepmind/mujoco_playground/blob/main/learning/notebooks/training_vision_1_t4.ipynb) of this tutorial, which builds + installs Madrona MJX from scratch on a cloud instance.

kevinzakka, 2025-04-06:
Going to close, @benbenwsh please let us know if you still have issues.

closed 2025-04-06 by kevinzakka