google-deepmind/mujoco_playground #77: Pretrained g1 policy doesn't seem to respect command well

dementrock, 2025-03-09:
I tried the play_g1_joystick.py example with the default g1 onnx policy, without connecting it to a gamepad so the command is always [0, 0, 0]. Seems like the robot still moves forward and doesn't track orientation well. Is this expected?

https://github.com/user-attachments/assets/7b535b02-8d63-4d88-9f44-1eb1bd12f4ad

kevinzakka, 2025-03-09:
Yes, this is not the best policy we trained. You'll notice for the Booster T1, we have a phase variable "hack" to encourage the policy to stay still. In the current joystick implementation for the G1, the reward terms are kinda fighting themselves at the 0 command (phase variable is encouraging feet to move while tracking term is encouraging the policy to stand still). Try the booster policy and check out the env implementation to see how to make the policy stand still.

kevinzakka, 2025-03-09:
Oh I actually left a comment [here](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/_src/locomotion/g1/joystick.py#L366-L371).

dementrock, 2025-03-09:
@kevinzakka I see, thanks! Do you plan to release an improved policy?

dementrock, 2025-03-09:
Just tried command [1, 0, 0] with vx=1. Seems like robot still deviates pretty quickly:

https://github.com/user-attachments/assets/4c2c52cc-05da-4e08-abb2-5b7f52e8a8db

So the issue doesn't seem limited to zero command?

kevinzakka, 2025-03-12:
Yes, I'll release one when I find some time, a bit slammed atm!

closed 2026-11-15 by btaba