google-deepmind/mujoco_playground #81: nan when training Humanoid with purejaxrl library

cvoelcker, 2025-03-14:
Dear team,

we have been trying to use the purejaxrl library in combination with mujoco_playground [1], but have run into an issue. On some dmc environments, most commonly humanoid, but also sporadically on walker, the position and velocity values of the environment seem to become nan. As far as we can tell this does not happen because of invalid actions, we have proper action clipping etc in place. We further guarded against any possible action issues by `action = jnp.nan_to_num(action, 0.0)` before passing the action to the environment.

Our changes to purejaxrl are minimal, we simply use the following wrapper to translate between mjx and gymnax

```
class MjxGymnaxWrapper(Environment):
    def __init__(
        self,
        env_or_name: str | MjxEnv,
        episode_length: int = 1000,
        action_repeat: int = 1,
        reward_scale: float = 1.,
        config: dict = None,
    ):
        if isinstance(env_or_name, str):
            if config is None:
                config = registry.get_default_config(env_or_name)
            else:
                config = ConfigDict(config)
            env = registry.load(env_or_name, config=config)
            if episode_length is not None:
                env = brax.envs.wrappers.training.EpisodeWrapper(
                    env, episode_length, action_repeat
                )
                env = BraxAutoResetWrapper(env)
            self.env = env
        else:
            self.env = env_or_name
        self.reward_scale = reward_scale
        super().__init__()

    def action_space(self, params):
        return gymnax.environments.spaces.Box(
            low=-1.0,
            high=1.0,
            shape=(self.env.action_size,),
        )

    def observation_space(self, params):
        if isinstance(self.env.observation_size, int):
            return Box(
                low=-float("inf"),
                high=float("inf"),
                shape=(self.env.observation_size,),
            )
        else:
            return Dict(
                {
                    key: Box(
                        low=-float("inf"),
                        high=float("inf"),
                        shape=(size,),
                    )
                    for key, size in self.env.observation_size.items()
                }
            )

    @property
    def default_params(self) -> gymnax.EnvParams:
        return gymnax.EnvParams()

    def reset_env(self, key, params=None):
        state = self.env.reset(key)
        return state.obs, state

    def step_env(self, key, state, action, params):
        action = jnp.nan_to_num(action, 0.0)
        state = self.env.step(state, action)
        return state.obs, state, state.reward * self.reward_scale, state.done > 0.5, {}
```

We have not observed this problem when training with brax ppo. However, the brax ppo implementation is rather hard to extend, which is why we would prefer working with other libraries.

[1] https://github.com/luchris429/purejaxrl

kevinzakka, 2025-03-14:
Hi @cvoelcker, this was flagged by @hartikainen as well. In fact, brax SAC has nan issues but brax PPO does not. I'm not sure if there's been a regression since the release (there's also been 3 mujoco updates since we ran these experiments) so we need to investigate.

cvoelcker, 2025-03-14:
Thanks for the quick reply! If this is a wider problem, I'll be patient :) Love what you did with the DMC overall, very nice new code!

fukangl, 2025-03-15:
Hi @cvoelcker @kevinzakka, I also facing a NaN issue when applying mass domain randomization to the Agility Digit-v3 humanoid robot. However, this issue does not appear in other humanoid robots like the Berkeley humanoid and T1. Digit robot has six rod (ball joint) for close chain constraints, which introduces six equalities constraints in the xml file. When I disable these equality constraints, the NaN issue will disappear. But we need to keep these constraints. 

I was wondering is there a way to resolve this while maintaining the constraints? Thanks!

```
rng, key = jax.random.split(rng)
dmass = jax.random.uniform(
    key, shape=(model.nbody,), minval=0.99, maxval=1.01
)
body_mass = model.body_mass.at[:].set(model.body_mass * dmass)
```

m-erdemm, 2025-03-17:
OOC, how did you manage to evaluate your trained purejaxrl policy in mujoco playground? For some reason, I get zero reward for my environments during eval, even though in trainings the policy gets expected rewards.

btaba, 2025-04-10:
Closing this issue due to inactivity. Any NaN issues related to potential regressions in JAX/MJ will be addressed here https://github.com/google-deepmind/mujoco_playground/issues/92

closed 2025-04-10 by btaba

rivergold, 2025-04-14:
> Hi [@cvoelcker](https://github.com/cvoelcker), a friend told me that it looks like newer JAX versions are causing the issues (as opposed to something on the MJX side). Can you check if `jax==0.4.35` resolves the NAN issues?

Hi @kevinzakka, sorry to bother you. 
Is this issue https://github.com/google-deepmind/mujoco_playground/issues/112 also caused by jax for MJX?

cvoelcker, 2025-04-15:
Hi, I also (finally) tested this, and got the same results. Changing the jax versions fixes nan issues in some envs, others, like humanoid, still produce nan. Tested with 0.5.3 and 0.4.35