google-deepmind/mujoco_playground #84: linvel in obs

kassasin, 2025-03-20:
I found in locomotion examples , observation contains "linvel" , I think it is very hard to get from imu . When I remove this item from observation, robot will be very hard to learn move forward . My question is "linvel" is necessary?

kevinzakka, 2025-03-22:
Hi @kassasin, linear velocity is definitely not necessary, but it improves tracking performance. I can confirm that we've successfully transferred policies without it.

kassasin, 2025-03-23:
> Hi [@kassasin](https://github.com/kassasin), linear velocity is definitely not necessary, but it improves tracking performance. I can confirm that we've successfully transferred policies without it.

Thank you kevin. Can I add acceleration of IMU and remove linear velocity in training stage . I noticed in a example of g1,  state contains noisy_linvel .
```
    state = jp.hstack([
        noisy_linvel,  # 3
        noisy_gyro,  # 3
        noisy_gravity,  # 3
        info["command"],  # 3
        noisy_joint_angles - self._default_pose,  # 29
        noisy_joint_vel,  # 29
        info["last_act"],  # 29
        phase,
    ])
```

kevinzakka, 2025-03-30:
Acceleration is typically pretty noisy. I think you can just remove the linear velocity term and tweak the reward function to improve the results!

closed 2025-03-30 by kevinzakka

kassasin, 2025-03-31:
> Acceleration is typically pretty noisy. I think you can just remove the linear velocity term and tweak the reward function to improve the results!

Thank you. I am working on this .