google-deepmind/mujoco_playground #91: Training log/feedback guidance?

jloganolson, 2025-03-30:
Are there any examples or guidance on the proper way to get more frequent logging during training akin to isaac lab (screenshot attached)

[image]

I'm finding that the default training is just minutes of no feedback, which isn't great when just getting started....

(replied to a [mujoco issue](https://github.com/google-deepmind/mujoco/issues/2441) with this question as well but thought i should x-post here)

kevinzakka, 2025-03-30:
Hi @jloganolson, you can actually do this. In the train function for PPO, use the following arguments:

```python
num_evals=0,
log_training_metrics=True,
```

kevinzakka, 2025-04-06:
Going to close, @jloganolson please let me know if the above was not sufficient.

closed 2025-04-06 by kevinzakka

jloganolson, 2025-04-09:
Sorry just getting back to this -- log_training_metrics doesn't appear to be an argument:
`train() got an unexpected keyword argument 'log_training_metrics'`

Confirmed (i think) with `print(inspect.signature(ppo.train))` not having that in the outpu

But I see it here in the brax repo: https://github.com/google/brax/blob/241f9bc5bbd003f9cfc9ded7613388e2fe125af6/brax/training/agents/ppo/train.py#L237

Any idea where the disparity comes from between the brax.train.train here and the one in the repo? 

Also confirmed I'm using brax==0.12.1

Any insight appreciated!
Logan

kevinzakka, 2025-04-10:
@jloganolson Doesn't hurt to just do `pip install git+ssh://git@github.com/google/brax.git`