google-deepmind/mujoco_playground #92: Running "humanoid_joystick.ipynb" reports an error

fushengcheng, 2025-03-31:
[image]
When training in the G1JoystickFlatTerrain environment, an error is reported as shown in the figure. How can I fix it?

kevinzakka, 2025-03-31:
I need more information. What GPU version? How many GPUs? Do you have a heterogeneous GPU setup (aka different GPUs on the same machine)? What JAX version?

fushengcheng, 2025-04-01:
> I need more information. What GPU version? How many GPUs? Do you have a heterogeneous GPU setup (aka different GPUs on the same machine)? What JAX version?

Thank you for your reply. The nvidia dirver version is 535.54.03 and CUDA version is 12.2. I have 2 homogeneous GPUs on the same machine. JAX version is 0.5.3.

Specialized123, 2025-04-14:
I have this problem too,I use nvidia 4090 24G,  NVIDIA-SMI:570.124.04,Driver Version: 570.124.04 ,CUDA Version: 12.8; JAX version : 0.5.3

fushengcheng, 2025-04-15:
> I have this problem too,I use nvidia 4090 24G, NVIDIA-SMI:570.124.04,Driver Version: 570.124.04 ,CUDA Version: 12.8; JAX version : 0.5.3

You can try using version 0.4.35 of JAX, which can solve this problem in some environments. However, when I attempted to incorporate the corresponding training environment of Go2 by referring to the joystick and getup environment of Go1, the getup environment still threw this error under the old version of JAX.

cvoelcker, 2025-04-15:
Hi, I also (finally) tested this, and got the same results. Changing the jax versions fixes nan issues in some envs, others, like humanoid, still produce nan. Tested with 0.5.3 and 0.4.35

42jaylonw, 2025-04-20:
> Hi, I also (finally) tested this, and got the same results. Changing the jax versions fixes nan issues in some envs, others, like humanoid, still produce nan. Tested with 0.5.3 and 0.4.35

I got same problem when randomizing `body_mass` in G1 environment, but these version solved it !

```
python==3.13.2
jax==0.4.38
jax-cuda12-pjrt==0.4.38
jax-cuda12-plugin==0.4.38
jaxlib==0.4.38
jaxopt==0.8.5
flax==0.10.4
brax==0.12.3
```

Hojunian, 2025-04-30:
I got the same problem when using rtx 4090, and solved the issue without changing JAX version.

I just added `jax.config.update('jax_default_matmul_precision', 'highest')`, following [https://github.com/google-deepmind/mujoco/issues/1484](https://github.com/google-deepmind/mujoco/issues/1484)

btaba, 2025-06-24:
After much digging, here's the fix https://github.com/jax-ml/jax/commit/ebba464b591fe2a8a4bdf6440672a78b471b22d2

We'll close the issue once there's a new JAX release

btaba, 2025-06-26:
Thanks @cvoelcker  for the details!

Since `jax==0.4.38` showed the same issue, the regression in this github issue is unrelated to the numerical issue you're seeing.

If you can try on a different GPU that would be awesome, to rule out the hardware. You can run in colab to get an A100 instance, it would be a great sanity check. Otherwise you can open a new issue with a minimal reproducible example, and we can take a look

btaba, 2025-08-13:
Closing this issue since MJX 3.3.4 was released

closed 2025-08-13 by btaba