google-deepmind/mujoco_playground #95: Regarding the setting of feedforward torque

Songchaopipi, 2025-04-01:
I hope to set feedforward torque in step() function, which is in the following code:

def step(
    model: mjx.Model,
    data: mjx.Data,
    action: jax.Array,
    n_substeps: int = 1,
) -> mjx.Data:
  def single_step(data, _):
    data = data.replace(ctrl=action)
    data = mjx.step(model, data)
    return data, None

  return jax.lax.scan(single_step, data, (), n_substeps)[0]

I noticed that the ctrl is the action and the action is the position target, does this mean it is a high-gain position control? If I want to set feedforward torque(such as solved by wbc), do I need to separately initialize a d.qfrc_applied and set it to the torque I calculated?

kevinzakka, 2025-04-01:
You can replace the position actuators in the XML with `<motor>` actuators and setting control will have semantics of torque.

Songchaopipi, 2025-04-01:
> You can replace the position actuators in the XML with `<motor>` actuators and setting control will have semantics of torque.

Thank you very much! I overlooked the settings in the XML file.

closed 2025-04-01 by Songchaopipi