google-deepmind/mujoco_playground #99: Quadruped?

yardenas, 2025-04-04:
Hi ant thanks again for the amazing work!

I have a [hacky implementation](https://github.com/yardenas/safe-sim2real/tree/main/ss2r/benchmark_suites/mujoco_playground/quadruped) of the deepmind control quadruped environment, currently only with the `Run` and `Walk` tasks.

I am happy to contribute a cleaner version of it to mujoco_playground, if that would be useful to anyone. Please let me know!

yardenas, 2025-04-11:
@btaba sounds good! I'm on it. It would probably take me a few weeks to start working on it and make the code cleaner. Hope to update you soon!

yardenas, 2025-05-04:
@btaba, two questions about the design of the environment:

1. Currently we do not sample the initial position of the quadruped as done in the original implementation as it slows down training (I think ~5 additional minutes). What's your opinion about that? I could try to optimize my implementation of _find_non_contacting_height, but not sure exactly if we can gain a lot.
2. I disabled some contacts of the knees with the ground to speed up performance. What do you think about that?

btaba, 2025-06-23:
Hi @yardenas , sorry for the late reply:

1. Instead of doing `_find_non_contacting_height` I would choose an initial position with noise that ensures none of the quadrupeds initially penetrate. We do this in robotic quadruped environments.
2. It's ok to disable contacts if they are not critical for the env behavior. Otherwise, check out [this](https://mujoco.readthedocs.io/en/stable/mjx.html#performance-tuning) (in particular broadphase section).

btaba, 2026-11-15:
feel free to add more DMC in the 1.0 Playground (we'll eventually get to it, hopefully)

closed 2026-11-15 by btaba