google-deepmind/mujoco_playground #109: G1 policy deployment questions

xkhainguyen, 2025-04-11:
Hi All,

Thanks for open-sourcing this awesome project!

I've been trying to reproduce the G1 locomotion results on real hardware but haven’t had a successful run so far. Meanwhile, I was able to deploy unitree_rsl_gym successfully. Hence, I have a few questions:

1. Your XML model seems slightly different—for example, it has separate Kp, damping, and armature parameters compared to unitree_rsl_gym's model. Did you perform system identification with the real hardware? 
2. Do PD gains in MJX have a different meaning compared to Isaac and hardware?
3. I've been tuning the rewards, but there are probably more factors to consider. Could you share your G1JoyStickFlatTerrain configurations (scales, noises, PPO parameters, etc.) and any insights on deployment?
4. I plan to remove linear velocity or anything requiring state estimation. This worked for unitree_rsl_gym with LSTM policies. Do you think this would work in this case as well?
I look forward to your response.

Many thanks,

kevinzakka, 2025-04-11:
1. No system identification in the true sense of the word.
2. PD gains on hardware match those in real. We're using <position> actuators which corresponds to sending position setpoints to the motors and the sim does a PD control loop under the hood.
3. I'll re-train and tune the G1 config we have soon and update the codebase (the one we deployed in real went through multiple iterations and so may have diverged). I'm busy with CoRL right now so can't give you a concrete date.
4. Yes, linear velocity is not strictly necessary.

xkhainguyen, 2025-04-12:
Thanks for your quick response!

Take the hip pitch joint for example:
1. How was the armature value of 0.00517752004 estimated?
2. Does it mean you sent a position setpoint, Kp = 75 and Kd = 2 to the Unitree G1 SDK?
2'. What are the advantages of having the PD loop (or postion actuators) in MJX? This is different from Isaac pipeline where PD output is computed in the rsl part.

3. Totally understood. Looking forward to the update.
4. Great!

jloganolson, 2025-04-30:
Re: retraining and tuning the G1 policy, it would be awesome if you could include some sort of log or run-through of what that involves (e.g. wandb sweeps? if so, using its optimization or just looking at the runs by hand? if so, what? etc).

jloganolson, 2025-05-02:
I got deployment stood up for the 23 dof (along with training and sim2sim) and isolated it in a new repo here:
(https://github.com/jloganolson/g1_23dof_basic_locomotion) 

Note: the policy using the current rewards (copied from 29dof with arms disabled) works [but is quite unstable](https://x.com/jloganolson/status/1918434795295793233)

xkhainguyen, 2025-05-03:
@jloganolson  

Thanks for your information. I had the same deployment issue last time. Looking forward to hearing from the developers.

xkhainguyen, 2025-08-11:
@kevinzakka  hope to hear more from you