google-deepmind/mujoco_playground #111: Modifying train rsl_rl for DeepMind Control Suite

0ifeng0, 2025-04-11:
Thank you for your excellent work on this open-source project.

I have configured rsl_rl_config, but it doesn’t seem to work with the DeepMind Control Suite environments. Could you please advise on how to modify train rsl_rl to support these environments? Your guidance would be greatly appreciated.

Thank you!

Best,

btaba, 2025-04-11:
Hi @0ifeng0 , indeed we don't have RSL-RL config set for DM Control Suite [here](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/config/dm_control_suite_params.py). I'm not an expert on RSL-RL config, you'll likely have to: [1] run a sweep and [2] base some params off of the brax one.

btaba, 2025-04-11:
@0ifeng0 let us know if you'd like to fix this!