google-deepmind/mujoco_playground #112: Brax trainning result is different with different jax version

rivergold, 2025-04-14:
I have trained 

```
python train_jax_ppo.py --env_name PandaPickCube
```

with different jax version:

1. jax=0.4.38：RL task is ok, final training reward `20152320: reward=1341.823`

https://github.com/user-attachments/assets/af1b1c0f-9726-4cb1-92da-2d2741e2b98f

2. jax=0.5.3：RL task is failed, final training reward `20152320: reward=708.486`

https://github.com/user-attachments/assets/9d895220-dfe4-434f-9e27-f8ab795edb1e

Why is the result so different with jax version……

ZaberKo, 2025-04-16:
Have a similar issue. Affect manipulation envs

btaba, 2025-04-16:
Hi @rivergold , indeed there seems to have been some kind of numerical precision regression in recent jax versions compared to 0.4.3* that we are still tracking down.

Generally speaking we cannot absolutely guarantee the same performance across versions, the better option (time permitting) is to redefine the rewards/environment to make the results a bit more robust to numerical diffs. Leaving the issue open

Related to: https://github.com/google-deepmind/mujoco_playground/issues/92

jloganolson, 2025-04-30:
I burned quite a bit of time (hours) troubleshooting what I thought were bugs in my rewards/mjcf until I randomly happened upon this issue. If this is going to persist for more than another week or two, suggest putting this in the readme.md akin to the nvidia GPU note. Happy to make a PR if helpful.

ZaberKo, 2025-05-02:
Does `export JAX_DEFAULT_MATMUL_PRECISION=highest` solve this issue?

jloganolson, 2025-05-02:
That line fixed the NaNs for me (did that step first per readme.md a few days earlier) but the humanoid locomotion policy only started working (the G1 could recover after being pushed) after switching to jax 4.3.8

btaba, 2025-06-24:
After much digging, here's the fix https://github.com/jax-ml/jax/commit/ebba464b591fe2a8a4bdf6440672a78b471b22d2

We'll close the issue once there's a new JAX release

cvoelcker, 2025-06-25:
Amazing catch! Thanks a lot