google-deepmind/mujoco_playground #124: LeapCubeReorient corrupts during inference

sizhe-li, 2025-05-11:
Thank you for the amazing simulation platform! I tried running `train_jax_ppo.py` with `--env_name LeapCubeReorient`. The `rollout.mp4` looks corrupt after one step; it seems to be related to the numerical stability of the simulator. Empirically, the cube and finger tips disappear after one step. Please see the video below.

https://github.com/user-attachments/assets/e3d97c67-a405-4664-950a-19d264ee881f

sizhe-li, 2025-05-11:
Update: it seems that jax=0.5.0 fixed the issue, I was using the latest version, which caused this problem. I'm not sure why.

closed 2025-05-12 by sizhe-li

yingxin666, 2025-05-30:
I also encountered this issue. At first, the loss was NaN, but switching to the jax=0.5.0 version you recommended resolved this problem. However, when testing the trained model later, it failed quickly, and the saved video was similar to yours. Do you have any insight into the cause of this problem?

Greenery-S, 2025-06-08:
I’ve run into the same issue using JAX 0.5.0 on a custom SoArm100 pick-and-place task. The reward suddenly becomes NaN in a seemingly unpredictable way—sometimes during training, sometimes only when evaluating, just as you described. 

After digging in, I discovered that the NaNs originate from the action argument passed into step(). It looks like there may be a bug in PPO’s network that occasionally produces invalid (NaN) actions.

[image]