google-deepmind/mujoco_playground #130: jaxlib._jax.XlaRuntimeError: INTERNAL: cuSolver internal error

PoplarCrystal, 2025-05-25:
Hello, I am excited for your work of RL.
I have tried mujoco-playground on several different PCs and GPUs, as several different versions

For the first time, (Apr, 1st, 2025)
I installed the mujoco_playground from source.
uv pip install -U "jax[cuda12]"
pip list show:
**jaxlib                    0.5.3                    pypi_0    pypi
mujoco-mjx                3.3.1                    pypi_0    pypi
nvidia-cublas-cu12        12.8.4.1                 pypi_0    pypi
nvidia-cuda-cupti-cu12    12.8.90                  pypi_0    pypi
nvidia-cuda-nvcc-cu12     12.8.93                  pypi_0    pypi
nvidia-cuda-runtime-cu12  12.8.90                  pypi_0    pypi
nvidia-cudnn-cu12         9.8.0.87                 pypi_0    pypi
nvidia-cufft-cu12         11.3.3.83                pypi_0    pypi
nvidia-cusolver-cu12      11.7.3.90                pypi_0    pypi
nvidia-cusparse-cu12      12.5.8.93                pypi_0    pypi
nvidia-nccl-cu12          2.26.2.post1             pypi_0    pypi
Nvidia-smi: 535**

OK, I run python train_jax_ppo.py, it works.

For the second time, (May 25th, 2025)
I do the same thing,
**jaxlib                    0.6.1                    pypi_0    pypi
mujoco-mjx                3.3.2                    pypi_0    pypi
nvidia-cublas-cu12        12.8.4.1                 pypi_0    pypi
nvidia-cuda-cupti-cu12    12.9.19                  pypi_0    pypi
nvidia-cuda-nvcc-cu12     12.9.41                  pypi_0    pypi
nvidia-cuda-runtime-cu12  12.9.37                  pypi_0    pypi
nvidia-cudnn-cu12         9.10.1.4                 pypi_0    pypi
nvidia-cufft-cu12         11.4.0.6                 pypi_0    pypi
nvidia-cusolver-cu12      11.7.4.40                pypi_0    pypi
nvidia-cusparse-cu12      12.5.9.5                 pypi_0    pypi
nvidia-nccl-cu12          2.26.5                   pypi_0    pypi
nvidia-nvjitlink-cu12     12.9.41                  pypi_0    pypi
nvidia-nvshmem-cu12       3.2.5                    pypi_0    pypi
Nvidia-smi: 550**

OK, I run python train_jax_ppo.py, it doesn't work.
the bug: 

 File "/home/XXX/anaconda3/envs/mujoco-mjx/lib/python3.11/site-packages/absl/app.py", line 261, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "/home/XXX/RL/mujoco_playground/learning/train_jax_ppo.py", line 353, in main
    make_inference_fn, params, _ = train_fn(  # pylint: disable=no-value-for-parameter
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/XXX/anaconda3/envs/mujoco-mjx/lib/python3.11/site-packages/brax/training/agents/ppo/train.py", line 385, in train
    env_state = reset_fn(key_envs)
                ^^^^^^^^^^^^^^^^^^
jaxlib._jax.XlaRuntimeError: INTERNAL: cuSolver internal error
--------------------
For simplicity, JAX has removed its internal frames from the traceback of the following exception. Set JAX_TRACEBACK_FILTERING=off to include these.

So, I don't know why?

dougfulop, 2025-05-25:
You need to use Jax 0.5.3.  Python 3.12.6 worked today along with earlier versions of cudnn.

PoplarCrystal, 2025-05-25:
你好，我已收到你的来信。若有重要事情，请短信告知！

Qazalbash, 2025-05-29:
The problem is the cublas version. Try to separately install it like `pip install --upgrade nvidia-cublas-cu12` and it should work [^1].

[^1]: reference: https://github.com/jax-ml/jax/issues/29042#issuecomment-2916978884

btaba, 2025-06-23:
Thanks folks for the input and for pinning down the issue. It looks like this was fixed on the jax side with 0.6.2.

closed 2025-06-23 by btaba