google-deepmind/mujoco_playground #134: body clipping through ground

milks69420, 2025-06-01:
The body of the models clip through the ground but the feet are still there

rlee3359, 2025-06-01:
To improve speed of the simulation (since collision is expensive), only feet contacts/collision are simulated. This is usually fine for walking/balancing and most tasks since you don't want the robot to be on the ground anyway. You can see this in the contacts defined in the xml for G1 (note feet only in the name):
https://github.com/google-deepmind/mujoco_playground/blob/ff0ea5629bd89662f6ffa54464e247653737ea45/mujoco_playground/_src/locomotion/g1/xmls/g1_mjx_feetonly.xml#L411
You can also see there are contacts defined for the hands against the thighs, which you can find reference to in the reward function, which is used to encourage the robot to not hit its hands against its thighs.

If you want to train a task like getting up off the floor, you will need to set collision for the body of the robot. For that I recommend checking the GO1 getup task which uses full collision version of the xml: https://github.com/google-deepmind/mujoco_playground/blob/ff0ea5629bd89662f6ffa54464e247653737ea45/mujoco_playground/_src/locomotion/go1/getup.py#L103

kevinzakka, 2025-06-01:
Perfect answer, thanks @rlee3359!