google-deepmind/mujoco_playground #137: Error

shaashou, 2025-06-06:
Traceback (most recent call last):
  File "/home/ygr/miniconda3/envs/mujo_rl/lib/python3.11/site-packages/jax/_src/xla_bridge.py", line 442, in discover_pjrt_plugins
    plugin_module.initialize()
  File "/home/ygr/miniconda3/envs/mujo_rl/lib/python3.11/site-packages/jax_plugins/xla_cuda12/__init__.py", line 230, in initialize
    _check_cuda_versions(raise_on_first_error=True)
  File "/home/ygr/miniconda3/envs/mujo_rl/lib/python3.11/site-packages/jax_plugins/xla_cuda12/__init__.py", line 199, in _check_cuda_versions
    _version_check("cuSPARSE", cuda_versions.cusparse_get_version,
  File "/home/ygr/miniconda3/envs/mujo_rl/lib/python3.11/site-packages/jax_plugins/xla_cuda12/__init__.py", line 139, in _version_check
    raise RuntimeError(err_msg) from e
RuntimeError: Unable to load cuSPARSE. Is it installed?
WARNING:2025-06-06 20:45:54,562:jax._src.xla_bridge:791: An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not installed. Falling back to cpu.
cpu

Greenery-S, 2025-06-06:
"but a CUDA-enabled jaxlib is not installed." 

Try this cmd `pip install "jax[cuda12]<0.5" jaxlib`, version 0.4.38 is quit stable.

btaba, 2025-06-23:
Resolving for now, I've had similar issues with jax installations, and have had to sometimes nuke all packages and re-install:

```
pip uninstall jax jaxlib jax-cuda12-pjrt jax-cuda12-plugin

pip install -U "jax[cuda12]" 
```

closed 2025-06-23 by btaba