google-deepmind/mujoco_playground #142: Madrona wrapper axes error due to a recent update

alexeiplatzer, 2025-06-11:
Wrapping an environment for vision using the madrona wrapper no longer works, the identity randomization function in the wrapper module fails for instance when doing a tree replace on the in_axes variable. With an error: `TypeError: Model.__init__() got an unexpected keyword argument 'light_directional'`. For an example see the training_vision_1 notebook. It no longer works despite previously working (yesterday). Maybe caused by an API change due to a recent update in mujoco. As a consequence, the required axes need an update as well.

When forcing the mujoco version to 3.3.2, everything works as previously, with 3.3.3 it doesn't.

btaba, 2025-06-23:
Hi @alexeiplatzer , we just updated to `light_type` which is a replacement of `light_directional` and bumped the playground version. Let us know if the issue persists?

btaba, 2025-06-23:
Closing the issue for now, but will re-open if there's still an issue. Please let us know