google-deepmind/mujoco_playground #153: Training performance collapse in HumanoidRun/Walk/Stand

cvoelcker, 2025-06-26:
Following up from #92 as @btaba suggested this is a separate issue.

The DMC Humanoid envs (run/walk/stand) result in nans in training after a while. Details from the other thread.

Info from the other thread

Hi, after doing extensive tests yesterday it seems like neither fix (concerning symmetric Cholesky decomposition) reliably solves the issue on our server. It does reduce the instances of it occurring though.

Specs are an single L40s GPU, Driver Version: 570.148.08, CUDA Version: 12.8, jax[cuda12]==0.6.2 (with manual fix), mujoco==3.3.3 (with manual fix), playground==0.0.5. I am running HumanoidRun, HumanoidWalk, and HumanoidStand from the DMC implementation shipped with mujoco_playground. I tested both fixes together and either in isolation. If necessary I can try to get some stats on how many runs fail with either fix. Somehow, HumanoidWalk seems to be the most affected.

Full list of jax related libraries installed

```
distrax==0.1.5
flax==0.10.6
gymnax==0.0.9
jax==0.6.2
jax-cuda12-pjrt==0.6.2
jax-cuda12-plugin==0.6.2
jaxlib==0.6.2
jaxopt==0.8.5
mujoco==3.3.3
playground==0.0.5
```

- Training for a long time (several million timesteps). Failures happen both with very good and bad policies, no clear difference. I am not running PPO but a custom algorithm, but it seems extremely stable in all other envs (brax and playground) so I am willing to bet that it is not an algorithm problem.
- Training ends with nans. I am specifically guarding against feeding nan actions to the environment (jax.nan_to_num), so as far as I can tell the source of the nans is the environment.
-  I tried roughly 30 runs per fix combination, all of them end in a failure at some point over a maximum of 200_000 steps with 1024 envs in parallel (so 200_000_000 timesteps in total)
-  It is as far as I can tell only HumanoidRun/Walk/Stand. I tested all playground DMC envs, and G1 and T1 Joystick Humanoid rough and flat terrain. Other DMC envs and the G1 and T1 environments seem fine.
-  jax==0.4.38 showed the same issues. I was unable to test jax==0.4.35 due to dependency conflicts with other libraries.
-  I did set the precision to highest. I can repeat the test on one of our 3090 GPUs.

I verified that this error is reproducible on an L40s and a 4090.

btaba, 2025-06-26:
Thanks for opening an issue, great to see you reproduced on 4090! Can you give us a way to reproduce?

cvoelcker, 2025-06-26:
I'll try to reproduce the issue with a clean purejaxrl implementation, that way I don't have to share current research code. I'm also still verifying some changes on the 4090, I'm double checking whether all library versions are correct.

cvoelcker, 2025-06-26:
Hi, I don't have a working brax setup. I made a minimal fork of purejaxrl in which I can reproduce the issue on the 4090 [link to repo](https://github.com/cvoelcker/purejaxrl). If you need to test this internally in g3, you only need the `purejaxrl/ppo_continuous_actions.py` and the `purejaxrl/wrapper.py` files, those should be easy to copy over. I also added a `requirements.txt` and `pyproject.toml` with exact library versions.

Let me know if you need me to set up the brax version as well.

cvoelcker, 2025-06-26:
Yes, sorry ^^, if you simply run `python purejaxrl/ppo_continuous_actions.py` the issue should arise very quickly.

cvoelcker, 2025-06-26:
The envs will first show nans and then a return value of exactly `return=4.611695766448975` which is the return of all 0 actions (because I catch the nans from the policy)

kevinzakka, 2025-06-26:
Hi folks, I can also help with debugging. Gonna test Baruch's fix on my own envs and see if I get NaNs as well and @cvoelcker will also try your fork.

cvoelcker, 2025-06-26:
BTW, the reason I am not using brax is that it is impossible to dig through the layers of config handling and indirection to find out what the actual policy implementation is. is there an easy way to figure this out? This would also really, really help with reproducing the results in the paper, which so far I have not been able to do with any other library. Let me know if I should open a separate issue for this.