google-deepmind/mujoco_playground #160: Unable to Reproduce MuJoCo Playground Paper Results (Longer Training, Lower Reward) with Unitree G1

s1lent4gnt, 2025-07-05:
Hi, thank you for this very cool project!

I successfully ran training using `humanoid_joystick.ipynb` with `G1JoystickFlatTerrain` env, with a few necessary fixes (submitted in PR #159). However, I'm unable to reproduce the results reported in the MuJoCo Playground paper.

**Setup:**

* Environment: Google Colab with A100 GPU
* Added `jax.config.update('jax_default_matmul_precision', 'highest')` to address numerical stability issues

**Issues Observed:**

* Training took **1 hour 30 minutes** wall-clock time, significantly longer than the **10 minutes** reported in the paper.
* The final reward is also noticeably lower than what's reported.

**Training Metrics:**

*Reward vs. Environment Steps:*[image]

*Reward vs. Wall-Clock Time:* [image]

Any suggestions or clarifications on how to better match the original results would be appreciated.

btaba, 2025-08-13:
Hi @s1lent4gnt , it looks like this was resolved? Let us know if there are any further questions. I will get the PR in, thank you for the fixes!

s1lent4gnt, 2025-08-13:
Hi @btaba, thanks for the reply!, yes every thing is clear for me now.