google-deepmind/mujoco_playground #172: Madrona-MJX

jiwon99yoon, 2025-07-29:
Hello, it's a really hot project and thanks for open-source this project

I tried some training on your open-source envs&codes but i got a issue

I wanna train vision-based PPO policy (in your env : PandaPickCube (--vision = True)),
But the brax, jax version dependencies are quite confused between mujoco_playground and Madrona MJX

so I wonder which version of brax, jax, you use in local computer to train that env.

thanks a lot

------
what i found

Brax 0.12.4 + JAX 0.5.3: error in cuSolver
JAX 0.6.0+: can't serve some API in Madrona-MJX(they said in github)
JAX 0.5.x: Dependency gap between Brax and CUDA (12.xx)

jonzamora, 2025-08-20:
Bumping this issue. @jiwon99yoon did you ever find a solution for this?

StafaH, 2025-08-21:
For madrona_mjx you will need jax<0.6.0. To avoid CUDA dependency issues, it might be necessary to install JAX through "jax[cuda12_local]"

I haven't tested with the different versions flying around, but i usually use jax==0.4.36 with CUDA 12.8 on my system. You will need to downgrade after installing mujoco-mjx, brax, playground.