google-deepmind/mujoco_playground #173: cube rotate simulation diverges in jax 0.7.0

edwhu, 2025-08-03:
Hello,
I installed mujoco playground following the readme, which ended up installing jax 0.7.0
Then I ran the cube rotate task using the `cube_rotate.ipynb`, and the hand's fingers fly away quickly:

https://github.com/user-attachments/assets/49df1995-851c-458b-83cc-988a3d118e4b

Downgrading to jax 0.5.0 resolves the issue, the fingers do not fly away.

https://github.com/user-attachments/assets/f829dc34-e64f-438a-8187-f8741b956d88

btaba, 2025-08-13:
Nice thank you for the follow-up!

edwhu, 2025-09-29:
It turned out that was an unrelated issue. Downgrading to 0.5.0 is the direct way to solve it. 

I think we also got it working with 0.7.0 but I don't remember the details on how we got it to work.