google-deepmind/mujoco_playground #178: Explanation of the done criteria in CartpoleBalance MJXEnv

vmstavens, 2025-08-14:
Hi, I am a Ph.d student doing reinforcement learning using MuJoCo and would like to test an algorithm of mine on the `CartpoleBalance` environment in the `dm_control_suite` in MuJoCo Playground. 

I am encountering an interesting problem where the environment never terminates, which seems to be caused by [this](https://github.com/google-deepmind/mujoco_playground/blob/d886c80e038b41c855287152c5dee381b3395796/mujoco_playground/_src/dm_control_suite/cartpole.py#L218) line, which to me does not make sense. How can `data.qpos` or `data.qvel` ever be NaN? Is this what really is implied or am i missing something? I have tried to `jax.debug.print` the values and my suspicion regarding the done flag never being `True` seems to hold.

Any an all insight is very much appreciated! :)

Andrew-Luo1, 2025-08-14:
Hi, code in the environment logic is not everything that causes termination. In the case of Cartpole, termination is meted by the training [wrapper](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/_src/wrapper.py#L87), when the specified episode length is hit.

The nan check is just good practice since the simulator can have numerical instability based on its settings.

kevinzakka, 2025-08-20:
Thanks @Andrew-Luo1 !

closed 2025-08-20 by kevinzakka