google-deepmind/mujoco_playground #201: JAX vs Warp backend training divergence in LeapCubeReorient

kywch, 2025-09-04:
Hello, 

I tried to replicate the JAX vs Warp training results on LeapCubeReorient using learning/train_jax_ppo.py with the PR https://github.com/google-deepmind/mujoco_playground/pull/200 applied. 

I saw that the Warp results are comparable to your results, but the JAX results were much higher. After seeing the njmax warning, I changed `njmax` to 200. For JAX, setting `njmax` to 128 (the original) still yielded high rewards. Here are the wandb logs: https://wandb.ai/kywch/mjxrl?nw=nwuserkywch

What training script did you use to get the results mentioned in the announcement https://github.com/google-deepmind/mujoco_playground/discussions/197?

Thanks!

[image]

btaba, 2025-09-04:
Hello @kywch , njmax and nconmax have no bearing on the JAX results. See reproducibility section [here](https://github.com/google-deepmind/mujoco_playground?tab=readme-ov-file#reproducibility--gpu-precision-issues). Let us know if you're able to find the diff (I'm assuming you used `train_jax_ppo` ?)

btaba, 2025-09-04:
also see https://github.com/google-deepmind/mujoco_playground/issues/177

kywch, 2025-09-08:
Thank you @btaba ! 

I read the preprint and saw that the jax results were reaching 400+. 

Fig 15:
[image]

Fig 16:
[image]

closed 2025-09-08 by kywch