google-deepmind/mujoco_playground #214: Booster T1 sim2real

edm-wang, 2025-09-20:
We've been trying sim2real deployment for a policy trained with the default playground script on a Booster T1 robot. 

Couldn't find any scripts in the repo so we tried to write our own. Comparing the playground "T1JoystickFlatTerrain" environment with Booster Gym, the main difference seems to be that Booster Gym does not use linear velocity as a part of their observation. We also found that Booster's low-level SDK does not provide linear velocity data, and only IMU data including RPY, Gyro and Accelerometer. Thus getting the linear velocity requires more state estimation setup on the deployment part. Booster Gym seems to support training a fully sim2real policy without needing linear velocity.

We were wondering if you'd be able to also open-source the Booster T1 deployment code. If that's too messy/not readily available, we'd be happy to write our own and open-source it, but look for your high-level guidance on whether we should do state estimation or let the policy handle that part. Interested to see if you guys tried Booster Gym and how it compares to playground. Since sim2real isn't the predominant focus of our project, we'd really appreciate any help/suggestions for any readily available pipelines and your experiences developing it.

Much appreciated in advance and hope y'all have a good one!

JanTempus, 2025-09-30:
Hello

So I have deployed a couple of policies trained with mujoco and mujoco-playground on the boosterT1. I simply removed noisy_linvel from the state and it can walk. It was quite janky when I deployed to the T1, but I did not do much fine tuning, so I don't know how good it can become. 

I wrote a driver in ROS2 such that the policy can communicate with the T1 and it works well, and is not too difficult.
I however set the kps to the following
6.0, 6.0,
6.0, 6.0, 6.0, 6.0,
6.0, 6.0, 6.0, 6.0,
45.0,
45.0, 45.0, 45.0, 45.0, 10.0, 10.0,
45.0, 45.0, 45.0, 45.0, 10.0, 10.0 
And the kd's to the following
 0.2, 0.2,
 0.5, 0.5, 0.5, 0.5,
 0.5, 0.5, 0.5, 0.5,
 5.0,
 5.0, 5.0, 5.0, 5.0, 3.0, 3.0,
 5.0, 5.0, 5.0, 5.0, 3.0, 3.0
  
Which is much lower than in the boosterT1 deploy code. When I set it to their values the robot was very erratic

edm-wang, 2025-11-05:
Thanks a lot for this! Will check it out and report our findings.

btaba, 2026-11-15:
Playground 1.0 allows native support for real robot deployment code, but I don't have access to a Booster (we only added YAM and SO101 for now). Plmk if someone wants to provide a demo, closing this for now though!

closed 2026-11-15 by btaba